ShadowLight8/Dynamic_RDS #4: Reduce FPP Restart For Settings Changes

iamwally, 2022-06-22:


ShadowLight8, 2022-06-23:
When changing a setting that can be dynamically updated on the transmitter, find a way to make that happen rather than having to restart FPP all the time.

ShadowLight8, 2022-06-23:
Might be able to use similar logic to https://github.com/ShadowLight8/Si4713_FM_RDS/blob/master/plugin_setup.php#L42 to trigger a reload.

ShadowLight8, 2022-07-19:
Working for audio settings. Looking to see if other settings should be able to be changed on the fly as well. Otherwise, doing a bit more testing before closing this out.

ShadowLight8, 2022-08-03:
In my testing, I've not found any other settings that need or make sense to be changed live. The mechanism is in place and it's quick to add other settings. We can open a new issue if needed for other settings.

closed 2022-08-03 by ShadowLight8