ShadowLight8/Dynamic_RDS #7: Some suggestions

Poporacer, 2022-07-16:
I took a quick look at this plugin, and it looks promising. Typically, the .php page for the plugin interface itself is named with the plugin name, so output.php should be named dynamic_rds.php. You don't have to, but that is the standard practice.
Also, you have the Plugin in the Inputs/Outputs tab, but the Inputs/Outputs tab typically refers to Pixel Data. It would be more appropriate to be in the Status/Control tab where other configuration settings are managed. But again, you can put it anywhere you want.

ShadowLight8, 2022-07-16:
This is great info! My goal is to align the plugin with the best practices and standards so I'll plan on making these changes. Thanks!

ShadowLight8, 2022-07-16:
Let me know if there are any other suggestions to better align with standard plugin practices.

ShadowLight8, 2022-07-19:
Closing for now, but if there are other suggestions, feel free to reopen.

closed 2022-07-19 by ShadowLight8