ShadowLight8/Dynamic_RDS #10: More information

focusedonsound, 2022-10-26:
I was trying to find another way of reaching out. 

I stumbled upon this project, and wanted more information as to if this is functioning? If so do you have more information on connecting this? I am thinking about purchasing one of these, and using this for this years light show.

Thank you

ShadowLight8, 2022-10-26:
Hi @focusedonsound! This is a great way to reach out. Yes, this project is functioning with a Raspberry Pi and the QN8066 board listed in the [readme.md](https://github.com/ShadowLight8/Dynamic_RDS/blob/main/README.md). I can help out with questions about connecting. The last picture in the readme.md shows the connection to the Pi, but it's not labelled. Let me know how I can help!

ShadowLight8, 2022-11-12:
Updated the [readme.md](https://github.com/ShadowLight8/Dynamic_RDS/blob/main/README.md) with annotated images and a bit more information. Feel free to reopen this issue if you have more questions. Thanks!

closed 2022-11-12 by ShadowLight8