ShadowLight8/Dynamic_RDS #20: RDS Engine Loop

ljhaydn, 2023-11-17:
Checking the logs, when the RDS engine fails - i get a constant loop of the below which doesn't ever recover or stop

18:10:51 INFO --- 2023-11-16
18:10:51 DEBUG Lock created
18:10:51 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:10:51 DEBUG Fifo already exists
18:10:51 DEBUG line INIT
18:10:51 INFO Processing init
18:11:36 INFO --- 2023-11-16
18:11:36 DEBUG Lock created
18:11:36 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:11:36 DEBUG Fifo already exists
18:11:36 DEBUG line INIT
18:11:36 INFO Processing init
18:12:21 INFO --- 2023-11-16
18:12:21 DEBUG Lock created
18:12:21 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:12:21 DEBUG Fifo already exists
18:12:21 DEBUG line INIT
18:12:21 INFO Processing init
18:13:06 INFO --- 2023-11-16
18:13:06 DEBUG Lock created
18:13:06 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:13:06 DEBUG Fifo already exists
18:13:06 DEBUG line INIT
18:13:06 INFO Processing init
18:13:51 INFO --- 2023-11-16
18:13:51 DEBUG Lock created
18:13:51 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:13:51 DEBUG Fifo already exists
18:13:51 DEBUG line INIT
18:13:51 INFO Processing init
18:14:36 INFO --- 2023-11-16
18:14:36 DEBUG Lock created
18:14:36 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:14:36 DEBUG Fifo already exists
18:14:36 DEBUG line INIT
18:14:36 INFO Processing init
18:15:21 INFO --- 2023-11-16
18:15:21 DEBUG Lock created
18:15:21 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:15:21 DEBUG Fifo already exists
18:15:21 DEBUG line INIT
18:15:21 INFO Processing init
18:16:06 INFO --- 2023-11-16
18:16:06 DEBUG Lock created
18:16:06 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
18:16:06 DEBUG Fifo already exists
18:16:06 DEBUG line INIT
18:16:06 INFO Processing init

ShadowLight8, 2023-11-17:
@ljhaydn Thanks for reporting an issue.

Could you also include a similar log snippet from Dynamic_RDS_callbacks.log that covers the same time frame. That will help me know what the Engine is trying to do.

ljhaydn, 2023-11-17:
Here's the log from the same time frame.  For additional information - I generally run an FSEQ animation (without music) and rely on After Hours Music during intermission of "off season" such as now during thanksgiving.  I do this so I can have variability in my lights even if a "show" isn't playing to keep it interesting for neighbors and people who stop by.

My current intermission sequences are 45 seconds and loop continuously.  I can see in the callbacks logs that it is resetting itself every 45 seconds.  But then causes the engine to either crash or loop with the previous message.

I did switch my scheduler to run them as effects and not a playlist - and that seems to have fixed the issue.  But I will have other playlists over christmas that may not have an audio sequence and I will rely on the after hours music plugin.

Thanks for the work on this plugin by the way!  Its awesome when the engine is running.  Far superior to the si4713 I have tried before.

18:10:51 INFO --- 2023-11-16
18:10:51 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:10:51 INFO Processing --type
18:10:51 INFO Engine restart detected, sending INIT
18:10:51 INFO Type playlist
18:10:51 INFO Playlist action playing
18:11:36 INFO --- 2023-11-16
18:11:36 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:11:36 INFO Processing --type
18:11:36 INFO Engine restart detected, sending INIT
18:11:36 INFO Type playlist
18:11:36 INFO Playlist action playing
18:12:21 INFO --- 2023-11-16
18:12:21 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:12:21 INFO Processing --type
18:12:21 INFO Engine restart detected, sending INIT
18:12:21 INFO Type playlist
18:12:21 INFO Playlist action playing
18:13:06 INFO --- 2023-11-16
18:13:06 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:13:06 INFO Processing --type
18:13:06 INFO Engine restart detected, sending INIT
18:13:06 INFO Type playlist
18:13:06 INFO Playlist action playing
18:13:51 INFO --- 2023-11-16
18:13:51 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:13:51 INFO Processing --type
18:13:51 INFO Engine restart detected, sending INIT
18:13:51 INFO Type playlist
18:13:51 INFO Playlist action playing
18:14:36 INFO --- 2023-11-16
18:14:36 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:14:36 INFO Processing --type
18:14:36 INFO Engine restart detected, sending INIT
18:14:36 INFO Type playlist
18:14:36 INFO Playlist action playing
18:15:21 INFO --- 2023-11-16
18:15:21 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:15:21 INFO Processing --type
18:15:21 INFO Engine restart detected, sending INIT
18:15:21 INFO Type playlist
18:15:21 INFO Playlist action playing
18:16:06 INFO --- 2023-11-16
18:16:06 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
18:16:06 INFO Processing --type
18:16:06 INFO Engine restart detected, sending INIT
18:16:06 INFO Type playlist
18:16:06 INFO Playlist action playing

ShadowLight8, 2023-11-17:
@ljhaydn Thanks for posting that and the additional information. I've started tracing the code and the Engine might be failing when reading the plugin config file. Can you post your plugin.Dynamic_RDS? You can find it in ~/media/config or can use the FPP Web Api (just swap in your FPP's IP) http://192.168.0.187/api/configfile/plugin.Dynamic_RDS

In the meantime, I'm going to add some code around the config file being read to kick out a useful error.

Glad to hear you like the plugin!

ljhaydn, 2023-11-18:
Here you go.  I've also been able to replicate it when it experiences an "audio error."  Say for instance, After Hours Music fails to stop before a playlist item with sound starts.  If FPP posts a sound error - the RDS plugin engine goes into the same unrecoverable loop and stops working.

DynRDSTransmitter = "QN8066"
DynRDSmpcEnable = "1"
DynRDSFrequency = "97.7"
DynRDSQN8066ChipPower = "122"
DynRDSPSStyle = "Visit|www.|lights|onfalcon|.com|{T}|{A}|"
DynRDSRTStyle = "Visit www.lightsonfalcon.com|{T}[ by {A}]|"
DynRDSQN8066AmpPower = "75"
DynRDSQN8066Gain = "5"
DynRDSEngineLogLevel = "ERROR"

ShadowLight8, 2023-11-18:
@ljhaydn I've tried what you have in your config and nothing seems wrong on my system. This might take a bit more back and forth to resolve, but I really want to get this working for you.

Next steps to try -
Update Dynamic_RDS to the latest - I pushed a few changes to capture exceptions in a few places so they will be included in the logs.
Also, change the Debugging Logging Level for Callback and Engine to DEBUG

Give it another try - I don't expect the behavior to change, but with the additional exception logging and the DEBUG level it might give me enough detail to fix. Please post both logs when you can.

ljhaydn, 2023-11-18:
Thanks Nick!  Message received.  I'll update and set to debug and get you some logs for both.

ljhaydn, 2023-11-18:
Engine failed again playing a sequence on loop with no audio track and After Hours running

Callback:
21:36:54 INFO --- 2023-11-17
21:36:54 DEBUG Arguments ['--type', 'playlist', '--data', '{"Action":"start","Item":0,"Section":"MainPlaylist","blankAtEnd":1,"blankBetweenIterations":0,"blankBetweenSequences":0,"currentEntry":{"deprecated":0,"enabled":1,"isFinished":0,"isPlaying":0,"isStarted":0,"note":"","playCount":0,"playOnce":0,"secondsElapsed":35,"secondsRemaining":185,"sequenceName":"Thanksgiving5-6.fseq","type":"sequence","videoOut":"--Default--"},"currentState":"playing","desc":"","loop":0,"loopCount":0,"name":"Thanksgiving5-6.fseq","random":0,"repeat":1,"size":1}']
21:36:54 DEBUG Checking for socket lock by /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:36:54 DEBUG Lock found - /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py is running
21:36:54 DEBUG Setting up write side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:36:54 DEBUG Fifo already exists
21:36:54 INFO Processing --type
21:36:54 INFO Type playlist
21:36:54 INFO Playlist action start
21:36:54 DEBUG Processing done
21:37:39 INFO --- 2023-11-17
21:37:39 DEBUG Arguments ['--type', 'playlist', '--data', '{"Action":"playing","Item":0,"Section":"MainPlaylist","blankAtEnd":1,"blankBetweenIterations":0,"blankBetweenSequences":0,"currentEntry":{"deprecated":0,"enabled":1,"isFinished":0,"isPlaying":1,"isStarted":1,"note":"","playCount":2,"playOnce":0,"secondsElapsed":0,"secondsRemaining":45,"sequenceName":"Thanksgiving5-6.fseq","type":"sequence","videoOut":"--Default--"},"currentState":"playing","desc":"","loop":1,"loopCount":0,"name":"Thanksgiving5-6.fseq","random":0,"repeat":1,"size":1}']
21:37:39 DEBUG Checking for socket lock by /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:37:39 DEBUG Lock not found
21:37:39 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:37:39 DEBUG Setting up write side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:37:39 INFO Processing --type
21:37:39 INFO Engine restart detected, sending INIT
21:37:39 INFO Type playlist
21:37:39 INFO Playlist action playing
21:37:39 DEBUG Processing done
21:38:24 INFO --- 2023-11-17
21:38:24 DEBUG Arguments ['--type', 'playlist', '--data', '{"Action":"playing","Item":0,"Section":"MainPlaylist","blankAtEnd":1,"blankBetweenIterations":0,"blankBetweenSequences":0,"currentEntry":{"deprecated":0,"enabled":1,"isFinished":0,"isPlaying":1,"isStarted":1,"note":"","playCount":3,"playOnce":0,"secondsElapsed":0,"secondsRemaining":45,"sequenceName":"Thanksgiving5-6.fseq","type":"sequence","videoOut":"--Default--"},"currentState":"playing","desc":"","loop":2,"loopCount":0,"name":"Thanksgiving5-6.fseq","random":0,"repeat":1,"size":1}']
21:38:24 DEBUG Checking for socket lock by /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:38:24 DEBUG Lock not found
21:38:24 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:38:24 DEBUG Setting up write side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:38:24 INFO Processing --type
21:38:24 INFO Engine restart detected, sending INIT
21:38:24 INFO Type playlist
21:38:24 INFO Playlist action playing
21:38:24 DEBUG Processing done
21:39:09 INFO --- 2023-11-17
21:39:09 DEBUG Arguments ['--type', 'playlist', '--data', '{"Action":"playing","Item":0,"Section":"MainPlaylist","blankAtEnd":1,"blankBetweenIterations":0,"blankBetweenSequences":0,"currentEntry":{"deprecated":0,"enabled":1,"isFinished":0,"isPlaying":1,"isStarted":1,"note":"","playCount":4,"playOnce":0,"secondsElapsed":0,"secondsRemaining":45,"sequenceName":"Thanksgiving5-6.fseq","type":"sequence","videoOut":"--Default--"},"currentState":"playing","desc":"","loop":3,"loopCount":0,"name":"Thanksgiving5-6.fseq","random":0,"repeat":1,"size":1}']
21:39:09 DEBUG Checking for socket lock by /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:39:09 DEBUG Lock not found
21:39:09 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:39:09 DEBUG Setting up write side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:39:09 INFO Processing --type
21:39:09 INFO Engine restart detected, sending INIT
21:39:09 INFO Type playlist
21:39:09 INFO Playlist action playing
21:39:09 DEBUG Processing done
21:39:54 INFO --- 2023-11-17
21:39:54 DEBUG Arguments ['--type', 'playlist', '--data', '{"Action":"playing","Item":0,"Section":"MainPlaylist","blankAtEnd":1,"blankBetweenIterations":0,"blankBetweenSequences":0,"currentEntry":{"deprecated":0,"enabled":1,"isFinished":0,"isPlaying":1,"isStarted":1,"note":"","playCount":5,"playOnce":0,"secondsElapsed":0,"secondsRemaining":45,"sequenceName":"Thanksgiving5-6.fseq","type":"sequence","videoOut":"--Default--"},"currentState":"playing","desc":"","loop":4,"loopCount":0,"name":"Thanksgiving5-6.fseq","random":0,"repeat":1,"size":1}']
21:39:54 DEBUG Checking for socket lock by /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:39:54 DEBUG Lock not found
21:39:54 INFO Starting /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_Engine.py
21:39:54 DEBUG Setting up write side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:39:54 INFO Processing --type
21:39:54 INFO Engine restart detected, sending INIT
21:39:54 INFO Type playlist
21:39:54 INFO Playlist action playing
21:39:54 DEBUG Processing done
21:40:39 INFO --- 2023-11-17
[... cut ...]
21:43:42 INFO PS ['Visit   ', 'www.    ', 'lights  ', 'onfalcon', '.com    ', 'Pentaton', 'ix - Kid', ' On Chri', 'stmas.mp', '3       ']
21:43:42 DEBUG RDSBuffer updateData
21:43:42 INFO RT ['Visit www.lightsonfalcon.com    ', 'Pentatonix - Kid On Christmas.mp', '3                               ']
21:43:42 DEBUG line MAINLISTThanksgiving5-6.fseq
21:43:42 INFO Processing MainPlaylist
21:43:42 DEBUG Playlist Name: Thanksgiving5-6.fseq
21:43:42 DEBUG Cleaning up fifo
21:43:42 INFO Exiting

[image]

ShadowLight8, 2023-11-18:
@ljhaydn Good news! I was able to duplicate the issue you're seeing and fix it. I just pushed the change, so you should be able to update Dynamic_RDS and have it run without the Engine loop issue. Give it a try and let me know if this resolves your issue.

What I learned is that when a single sequence is directly run from the scheduler or status page, FPP sets the playlist name as <sequence name>.fseq. Engine would use the api/playlist/<sequence name>.fseq to determine the playlist length, but it would return "". This caused a parsing error which would stop the Engine. Next time the sequence started, Callback sees that Engine isn't running and restarts it.

The solution was to check the playlist name for a '.' since that isn't a valid character in a regular playlist. If it has a '.', then the playlist_length is 1, otherwise it will use the api to get the length.

ljhaydn, 2023-11-18:
@ShadowLight8 - I really appreciate you taking the time to look into this and how quickly you were able to find a fix!  I did some initial testing this morning, and everything appears to be working as intended - no more loops or engine failed.  As I switch over to my more complex scheduling after thanksgiving, I'll re-open this bug.

Thanks again!  And happy holidays!

closed 2023-11-18 by ljhaydn

ShadowLight8, 2023-11-18:
You are most welcome @ljhaydn! Thank you for reporting the issue and providing the various logs to help me track it now. Happy Holidays!

PR #22 (2023-11-18): Playlist length fix for single sequence
Fixes #20 
When a single sequence is directly run from the scheduler or status page, FPP sets the playlist name as <sequence name>.fseq. Engine would use the api/playlist/<sequence name>.fseq to determine the playlist length, but it would return "". This caused a parsing error.

Solution was to check playlist name for a '.' since that isn't a valid character in a regular playlist.