ShadowLight8/Dynamic_RDS #24: Custom interconnect cable parts

chrkov, 2023-12-12:
Do you have a list of the parts needed for the custom cable between the Pi and board?   Mainly need the plug for the transmitter you recommend.

iamwally, 2023-12-13:
It depends on people’s capabilities what I would recommend.  Some people are very comfortable with crimping and making connectors some are not.  You could go this way:
Kidisoii XH 2.54mm JST Connector with 22AWG Pre-Crimped Wire,JST XHP-5,2/3/4/5/6/7/8/9/10/12 Pin Housing JST Adapter Cable Connector with Length of 10/20/50/80/100cm Wire (2 Sets XH2.54 5Pin -20cm) https://a.co/d/9qUz3iG

Which has the wires precrimped.

Or you could go this way:

Hilitchi 560Pcs 2.54mm 2/3/4/5 Pin Housing and Male/Female Pin Head Connector Adapter Plug Set Perfectly Compatible with JST-XHP https://a.co/d/4X7qpTB

and do all the crimping yourself.

Same goes for the other side:

QLOUNI 620Pcs 2.54mm Pitch JST SM... https://www.amazon.com/dp/B0774NMT1S?ref=ppx_pop_mob_ap_share

I used the 6 pin connectors from this kit

And then I bought these wires 

EDGELEC 120pcs 30cm Dupont Wire... https://www.amazon.com/dp/B07GD1W1VL?ref=ppx_pop_mob_ap_share

 just changed housings and crimped the JST on other end so it saved me crimping on one end.

Hopefully some of this makes sense.

chrkov, 2023-12-13:
That information is exactly what I needed and made complete sense.  

Thank You!

closed 2023-12-17 by ShadowLight8