ShadowLight8/Dynamic_RDS #37: Dynamic RDS does not restart with FPPD Restart.

chrkov, 2024-01-15:
This is a new one.   When I boot FPPD, everything starts up just fine and runs.   But if/when I do a FPPD Restart (not reboot), dynamic RDS starts and then stops.   Here is a log set to excessive of it happening.

21:21:11 INFO --- 2024-01-14
21:21:11 DEBUG Lock created
21:21:11 DEBUG Setting up read side of fifo /home/fpp/media/plugins/Dynamic_RDS/Dynamic_RDS_FIFO
21:21:11 DEBUG Fifo already exists
21:21:11 DEBUG line INIT
21:21:11 INFO Processing init
21:21:11 INFO Config {'DynRDSEnableRDS': '1', 'DynRDSPSUpdateRate': '4', 'DynRDSPSStyle': 'Merry|Christ-|  -mas!|{T}|{A}|[{N} of {C}]', 'DynRDSRTUpdateRate': '8', 'DynRDSRTSize': '32', 'DynRDSRTStyle': 'Merry Christmas!|{T}[ by {A}]|[Track {N} of {C}]', 'DynRDSPty': '2', 'DynRDSPICode': '3DC6', 'DynRDSTransmitter': 'QN8066', 'DynRDSFrequency': '88.9', 'DynRDSPreemphasis': '75us', 'DynRDSQN8066Gain': '0', 'DynRDSQN8066SoftClipping': '0', 'DynRDSQN8066AGC': '0', 'DynRDSQN8066ChipPower': '122', 'DynRDSQN8066PIHardwarePWM': '1', 'DynRDSQN8066AmpPower': '100', 'DynRDSStart': 'FPPDStart', 'DynRDSStop': 'Never', 'DynRDSCallbackLogLevel': 'INFO', 'DynRDSEngineLogLevel': 'EXCESSIVE', 'DynRDSmpcEnable': '0', 'DynRDSAdvPISoftwareI2C': '1', 'DynRDSAdvPIPWMPin': '12,4', 'DynRDSQN8066DigitalGain': 0, 'DynRDSQN8066InputImpedance': 1, 'DynRDSQN8066BufferGain': 1}
21:21:11 INFO Using i2c bus 1
21:21:13 DEBUG RDSBuffer init
21:21:13 DEBUG RDSBuffer updateData
21:21:13 INFO PS ['        ']
21:21:13 DEBUG RDSBuffer init
21:21:13 DEBUG RDSBuffer updateData
21:21:13 INFO RT [' \r']
21:21:13 INFO New RDS Data
21:21:13 DEBUG RDS Values {'{T}': '', '{A}': '', '{N}': '', '{L}': '', '{C}': ''}
21:21:13 DEBUG RDS Data [Merry   Christ-   -mas! ]
21:21:13 DEBUG RDS Data [Merry Christmas!                ]
21:21:13 DEBUG QN80xx updateRDSData
21:21:13 DEBUG RDSBuffer updateData
21:21:13 INFO PS ['Merry   ', 'Christ- ', '  -mas! ']
21:21:13 DEBUG RDSBuffer updateData
21:21:13 INFO RT ['Merry Christmas!                ']
21:21:13 INFO Starting QN80xx transmitter
21:21:13 EXCESSIVE I2C read at 0x06 of 1 byte(s) returned 0x34
21:21:13 EXCESSIVE I2C write at 0x00 of 0xe3
21:21:13 EXCESSIVE I2C write at 0x02 of 0x10
21:21:13 EXCESSIVE I2C write at 0x07 of 0xe8 0x0b
21:21:13 EXCESSIVE I2C write at 0x19 of 0x22
21:21:13 EXCESSIVE I2C write at 0x1b of 0x42
21:21:13 EXCESSIVE I2C write at 0x01 of 0x41
21:21:13 EXCESSIVE I2C write at 0x00 of 0x0b
21:21:13 EXCESSIVE I2C write at 0x24 of 0xb8
21:21:13 EXCESSIVE I2C write at 0x24 of 0x38
21:21:13 EXCESSIVE I2C write at 0x27 of 0x3a
21:21:13 EXCESSIVE I2C write at 0x6e of 0xb7
21:21:13 EXCESSIVE I2C write at 0x28 of 0x11
21:21:13 DEBUG Setting up PWM
21:21:13 DEBUG Setting PWM period to 18300
21:21:13 DEBUG Cleaning up fifo
21:21:13 DEBUG Stopped PWM
21:21:13 INFO Disabled PWM
21:21:13 INFO Exiting

ShadowLight8, 2024-01-15:
At a quick glance tonight, I made some changes to support using either hardware PWM, but I failed to make similar changes in the clean up code. Looks like I need to fix that and add some additional logging.

closed 2024-01-15 by ShadowLight8

ShadowLight8, 2024-01-15:
@chrkov Fix has been merged. Another good catch! Feel free to reopen this if you still have issues.

chrkov, 2024-01-15:
Works perfect now.  Thanks!

Christopher Kovacik

***Notice: This email may contain privileged or confidential information and is for the sole use of the intended recipient(s). If you are not the intended recipient, any disclosure, copying, distribution, or use of the contents of this information is prohibited and may be unlawful. If you have received this electronic transmission in error, please reply immediately to the sender that you have received the message in error, and delete it. Thank you.