ShadowLight8/Dynamic_RDS #39: Get Track number from FPP playlist instead of audio file tag

chrkov, 2024-01-17:
Right now,
the track number {N} is obtained from the audio file.  This works fine.
the item count {C} is obtained from FPP Main Playlist section.  This also works fine.

Can we get a Value for the position in the Main Playlist section?   

The problem with the existing setup is two fold:
1.  If you change the position of a sequence in the playlist, you have to update the audio file tag.
2.  If you have the same sequence in two different playlists, and in two different positions, one has to be wrong.

Not sure if this is even possible because you need hooks into FPP to know the position, but if possible, a new value (  {P} maybe?  ) that will tell the position in the Main Playlist would work well.

ShadowLight8, 2024-01-17:
I think this is possible. It is similar to #19 so I'll look into adding this.

ShadowLight8, 2024-01-18:
Done! PR #40 is up and I'm testing the changes.

chrkov, 2024-01-18:
What is the variable?  Nothing updated in the Text Guide.   Did you use {P}?

ShadowLight8, 2024-01-18:
I just finished checking things and merged to main. You can update the plugin again. Yes {P} it is :)

ShadowLight8, 2024-01-18:
@chrkov Let me know if this works for you. The Playlist Position counts any item in the Main Playlist section. If you have Pause or other items in that section, they get counted too.

chrkov, 2024-01-18:
I just tested it and it works perfectly.   Very Nice!   you knocked that out quick....  I have a couple other things I will post up.   Not sure if they are possible or not, but cant hurt to ask.

ShadowLight8, 2024-01-18:
@chrkov Bring it! I like a challenge :)

ShadowLight8, 2024-01-18:
Closing out

closed 2024-01-18 by ShadowLight8