ShadowLight8/Dynamic_RDS #41: Integrate into MQTT

chrkov, 2024-01-18:
I use MQTT to monitor everything for the show.   Nice to have a single dashboard to see the status of everything.

Is it possible to add/send these settings using MQTT that the pi is already doing?

PS RDS (Send what is being sent to transmitter)
RT RDS (Send what is being sent to transmitter)
Pi Code
Frequency 
Gain 
Enable Soft Clipping 
Enable AGC 
Chip Power 
Enable PWM 
Amp Power 
MPC Support
Use PI Software I2C

ShadowLight8, 2024-01-19:
I've not done a ton with MQTT, so it might take a bit to figure out. When you say you use MQTT for monitoring, what are you using for a dashboard? Would be good to have that for testing.

chrkov, 2024-01-19:
I use Home Assistant which also runs the mosquito MQTT broker.

Sample
[image]

chrkov, 2024-01-20:
If you have any questions on MQTT layout or options, let me know.   I am pretty proficient with it and understand the workings of it.

ShadowLight8, 2024-01-30:
Just notes as I'm researching (feel free to comment!):

- Not installed by default, but at first enable: pip install paho-mqtt
- Does plugin need connection settings or just use FPPs?
- Ref - https://github.com/FalconChristmas/fpp/blob/master/docs/MQTT.txt
-- Topics like {optional prefix}/falcon/player/{hostname}/{sub_topic}
-- ? {optional prefix}/falcon/player/{hostname}/Dynamic_RDS/{sub_topic}
-- sub_topics needed: status (looking at fppd_status as an example)
- Update status any time RDS data is updated (new title, artist, etc)
- Flat list of key:value pairs or hierarchical layout (basically the config groupings like RDS settings, Transmitter Type, Audio, etc)
- More or less taking the current config, plus current data and sending to MQTT

chrkov, 2024-01-31:
Here is my 2 cents on some of it.

Does plugin need connection settings or just use FPP's - My though would be to piggie back off FPP's MQTT connection settings and just go into a subtopic under what is set.  I cant think of a single reason why you would want to connect to a different MQTT server than what FPP is connecting to.

Update status any time RDS data is updated - Yes..  I would say only update when anything changes.   I would recommend adding a status topic showing the plugin is running (FPP does a 'ready' topic for this.  Update that if the plugin shuts down and maybe also set a will on it in case of crash (they just added the will option in fpp 8).

Flat list of key:value pairs or hierarchical layout - My thoughts would be key:value pairs under one subtopic.  Another thought would be to do a JSON topic (i.e. dynamic_rds_status) like what fppd_status does now.   There are pros and cons to both.  But seeing that FPP does that now for the majority of its settings, maybe staying with that format makes the most sense?   Plus maybe you could copy part of the code over to help?  Just thoughts.   I personally like the JSON option just because it would be consistent.

ShadowLight8, 2024-02-04:
I've been making some good progress.

All the config screen and settings stuff is done. Including using the built in update popup window that FPP uses (like when installing the plugin) for installing paho-mqtt.

Basic connection settings are in place, like host, post, topic prefix, username, password. I've not tried to deal with Client ID or CA File yet. Not sure how much they get used, but I've seen a bit about them in the paho docs.

I have the ready topic updating 1 at startup, 0 at shutdown, -1 if something went wrong. I also have a test message going to status when the RDS data updates. I decided to go with 'falcon/player/FPP-DevPi/plugin/Dynamic_RDS/*' at the topics. This seems to better match with the pattern of other FPP topics.

Next up will be pulling the current config information and live data to pack into a json message to the status queue.

ShadowLight8, 2024-02-04:
Initial version is looking good. I ended up splitting the live RDS data and the plugin config into two topics. They update at different times and rates, so I think this is will work better. I put the notes on Pull Request #47 about the 3 topics.

It's been a while since I've tried it, but you might be able to manually install the plugin from the mqtt_status branch using:
https://raw.githubusercontent.com/ShadowLight8/Dynamic_RDS/mqtt_status/pluginInfo.json

If not, I'm running it right now and so long as nothing blows up, I'll push to main later today.

@chrkov Let me know if this covers what you were looking for, any feedback about how it's working, and/or if changes are needed.  😄

ShadowLight8, 2024-02-11:
I'm going to go ahead and merge in the MQTT changes, along with the other random bug fixes in that branch. I've run a scheduled playlist with after hours music in between without issue. I've only used the mosquitto_sub tool for testing, so there could be other things I'm missing. I'm good with making additional changes to support MQTT, so I'll keep this issue open.

ShadowLight8, 2024-02-11:
Reminder that I need to include documentation about this on the main readme.md

chrkov, 2024-02-11:
Ok.   Installed this version, but things are not looking good for me.   

1.  I am not getting any RDS output to the radio when I can get the service running.   
2.  If I restart FPP, the plugin will no longer start.  I have to reboot FPP to get it to get it running.  I get the below error.
[image]
Running i2cdetect does see it though.
[image]

I do see the MQTT stuff and from what I see, looks good.  I will play with that more this week.

Attached is the support file.
[Dynamic_RDS_logs_config_20240211140352.zip](https://github.com/ShadowLight8/Dynamic_RDS/files/14233838/Dynamic_RDS_logs_config_20240211140352.zip)

chrkov, 2024-02-12:
Scratch comment 1.   Turns out my test RDS radio needed to be rebooted and that was the issue.

ShadowLight8, 2024-02-12:
Ok - I'll look over the logs tonight. Glad to see the download button it working 😄

chrkov, 2024-02-20:
That seems to have fixed it.  I don't have any issues when restarting FPP now.

ShadowLight8, 2024-02-21:
@chrkov Awesome! If there's anything else needed on the MQTT side, let me know and I can make the tweaks or changes. I'm going to close this issue, but we can reopen if needed.