ShadowLight8/Dynamic_RDS #42: Display status on OLED screen

chrkov, 2024-01-18:
Most of my Pi's and BBB's have OLED screens cause it makes it so much easier to see the status' when by the equipment.

Not sure if this is even possible with FPP, but can you push info onto the OLED status screen also.   

I simple blurp maybe showing Dynamic RDS is running?    Maybe as simple as Freq / Power ?

ShadowLight8, 2024-01-19:
Which OLED screens are you using on the Pi's? This would give me another excuse to buy more hardware 😆 

I'll have to poke around FPP to see if they share control of it or provide a way to add to what they display.

chrkov, 2024-01-19:
A bit of a mixture.  If the Pi is powered by POE:
https://www.amazon.com/gp/product/B07Z1Y47C7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

If not POE:
https://kulplights.com/product/pi-oled/

and on the BBB's those are pretty much all connected to kulps and the oled is on the controller.

kgreene1983, 2026-09-20:
I would like this as well

ShadowLight8, 2026-09-20:
> I would like this as well

Let me know what type of display you're using or interested in. Thanks!

kgreene1983, 2026-10-02:
just a 0.96 Inch OLED Display Module, 128x64 Pixel I2C SSD1306 Luminous OLED Screen Board Compatible with Arduino Raspberry Pi ESP32

it work with fpp without any config... would be good if showed - FM freq, RDS Data, and if active