ShadowLight8/Dynamic_RDS #45: Investigate using lifecycle events from FPP

ShadowLight8, 2024-01-28:
I learned that plugins can register for the FPP lifecycle events of "start" and "shutdown". A restart is a "shutdown" followed quickly by a "start".

The "start" event is already covered by the "list" call to callback.py, so nothing new would be needed.

The "shutdown" event would be new code to allow for a clean shutdown of the engine. Mostly this will ensure that hardwarePWM is stopped and softwarePWM is set to a low state. Currently, when FPP stops, it kills child processes (like engine.py), so PWM can be left in an unexpected state (such as the transmitter being on at full power, which gets very HOT).

This means the engine.py will start, shutdown, and restart along side of FPP. Whereas today, it doesn't quite do that (especially around restarts).

@iamwally @chrkov @ThomasRa1 - Any concerns with this change?

iamwally, 2024-01-28:
No actually that seems better and how it should be to me😎
John Maugeri
***@***.***

chrkov, 2024-01-28:
I agree.   This sounds like the way it should run.

closed 2024-01-29 by ShadowLight8