ShadowLight8/Dynamic_RDS #51: Compatibility with FPP 8.0

cbuxton5200, 2024-09-17:
Attaching the log file from my controller which has been reflashed using FPP 8.0.  I had to manually configure the dtoverlay setting to get it this far, but it is still logging errors.
[Dynamic_RDS_logs_config_20240916192056.zip](https://github.com/user-attachments/files/17021316/Dynamic_RDS_logs_config_20240916192056.zip)

ShadowLight8, 2024-09-17:
Thanks for letting me know and using the log/config zip tool :) I'll take a look so see what's up.

I'm guessing there are a few tweaks need to be fully compatible with 8.0. I'd been testing against beta builds previously.

ShadowLight8, 2024-09-17:
@cbuxton5200 If you set the PWM power to 0, do things start working again? The most recent errors in your logs look similar to PWM interfering with the I2C communications.

ShadowLight8, 2024-09-25:
Merged #52 to improve FPP 8 compatibility

cbuxton5200, 2024-11-15:
This is working great under FPP8, thank you!

closed 2024-11-15 by cbuxton5200