ShadowLight8/Dynamic_RDS #55: Fresh Install of FPP 8.5.1 and cannot get engine to run at all (Could be me! LoL)

iamwally, 2025-06-07:
Fresh install of FPP 8.5.1, fresh install of plugin and no PWM or external power to transmitter, so simplest setup and I cannot get the engine to run.
Hardware is detected on i2c so pi sees the transmitter but no engine running. tried to switch back and forth between software i2c and hardware and it seems the checkbox isn't working as I get a message that hardware i2c still seems active.  Right now I am at a complete loss as to why I cannot get this working in this simplest setup?!

[Dynamic_RDS_logs_config_20250111194348.zip](https://github.com/user-attachments/files/20638328/Dynamic_RDS_logs_config_20250111194348.zip)

ShadowLight8, 2025-06-07:
Is this with a Raspberry Pi 5?

iamwally, 2025-06-07:
Nope Raspberry Pi3b+

ShadowLight8, 2025-06-12:
In the /boot/firmware/config.txt is dtoverlay=vc4-kms-v3d. One element this adds is an i2c bus for DDC/EDID communication on the HDMI ports. On a Raspberry Pi 4, there are 2 HDMI ports, so this overlay adds an i2c-20 and i2c-21 bus. However, on a Raspberry Pi 3B+ or other single HDMI port variant, it add i2c-2. While the code on the plugin config page guarded against using i2c-2 unless on a BBB, the code in basicI2C.py did not. This was fixed in #56 and should prevent single HDMI port Pis from using i2c-2 in error.

I think this commit in FPP is where the change or update was introduced around this overlay:
https://github.com/FalconChristmas/fpp/commit/936ad3e307359adbeeca528eb11c8a92a5133803#diff-9a685a5f265e4edb9aee177637186f32e2568df7ad9093223bef3b6a1b88ce0f